nettings/medianet #50: jackd hangs with JackAudioDriver::ProcessGraphAsyncMaster process error after boot

nettings, 2020-05-15:
then recovers after > 1min.
May 15 21:42:26 mn-Dantax jackd[643]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
May 15 21:42:26 mn-Dantax jackd[643]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
May 15 21:42:26 mn-Dantax jackd[643]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
[repeats hundreds of times, many per second]

Hardware is Pi4 with DAC+ADC with AMP60 on top, kernel 4.19.97

System software: Raspbian GNU/Linux 10 (buster)
medianet overlay: Tue May 5 23:02:49 2020 +0200 b13c3ad897a5ad9a535dcf33b91de23c9604dddc

nettings, 2020-05-15:
not 100% reproducible, happened two out of three boots so far.

nettings, 2020-07-11:
This was only ever seen on several DAC+ADC, possibly also without AMP60. No further incidents in a while, all machines had a routine kernel update. Needs to be watched, though.

nettings, 2020-09-09:
same issue on AMP4/RPI4
Linux mn-vp200a-l 5.4.51-v7l+

nettings, 2020-09-10:
Usually not self-recovering. Killing mod-host helps most of the time, but not always.

nettings, 2020-09-10:
Funny: is this the race condition I'm seeing? There is a jack_lsp job running in parallel...
```
Sep 10 22:16:10 mn-vp200a-l zita-n2j[646]: Connect: can't connect named futex name = jack_sem.1001_default_lsp-01 err = No such file or directory
```
This was the first time such a message occurred.

nettings, 2020-09-11:
It looks like jack client creation and/or jack_lsp is racy and can send jackd into this death spiral. After reintroducing waiting for ports, the issue seems to have disappeared or at least become so rare that I haven't observed it since. Needs to be looked at further. See commits c214689c000b4d004db45d879592da1785c1c445 and 4f551386a5fbb22ccb87d058dbcf624916a1c576 .

nettings, 2021-03-31:
It still happens when the boot process is stalled because of an fsck or because systemd-timesyncd stalls and then syncs, at which point all Jack clients are long ready and spinning on mn_jack_wait or mn_jack_connect, which seems to trigger the death spiral.

nettings, 2022-08-14:
Not observed since.

worikgh, 2023-01-03:
This is happening to me on a Raspberry Pi 4, Pisound card, and PatchboxOS

daemon.log filling up with lines like:

`Jan  3 15:35:48 patchbox jackdrc[19278]: JackAudioDriver::ProcessGraphAsyncMaster: Process error`

nettings, 2023-11-26:
again observed several times on bookworm, only ever with the mod-monitor client.